xreef/PCF8575_library #1: Esp32 - changing pins

WTHfloyd, 2019-12-20:
I am trying to add a PCF8575 (well actually a PCF8574) to an esp32cam . But should be the same...
(note I am a little new at this...)

I am trying to map to different pins for the SDA and SCL - since the esp32cam SDA and SCL pins are not avaialbe. But I believe I should be able to map them to most any pin.
But can't make it work. 

I wanted to change pins for the SDA and SCL on an esp32.
I wanted to use pins 27 and 14 for an esp32 for testing (but really I wanted this for an esp32cam using pins 12 and 2).

I tried using the wire.h library and the PCF8574 lib...but for some reason it keep son forcing pins 21 and 22 as the SDA and SCL...and I did read that I need to sort out the .cpp files to make sure it is not being forced to set SDL and SCL pins... on this tutorial https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/ but I have no clue how to do that!!

And I see on a different sensor - the libraries will allow you to change pins...with the twowires...

"In those cases, you need to take a closer look at the .cpp library files and see how to pass your own TwoWire parameters."  https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/   .  But I cannot figure out how to do this...

For testing I was using the code below.
>>> using I2C scanner I get address for PCF as 0x38 
>>> running below sketch - SDA and SCL are still using pins 21 and 22 ...and it works - the LED blinks on/off using pins 21 and 22...

But pins should be mapped to 27 and 14....
How to fix??
Can you update library to allow mapping?? Or am I doing something wrong?
Thanks!

***

#include "Wire.h"
#include "Arduino.h"
#include "PCF8574.h"

TwoWire I2Cone = TwoWire(0);

 PCF8574 pcf8574(&I2Cone, 0x38);

(note your tutorial it think it has typo... line 39 >>  PCF8574 pcf8574(&amp;I2Ctwo, 0x20); >>> I think it should be PCF8574 pcf8574(&2Ctwo, 0x20);  ) Your editor is added letters...

void setup(){

 Serial.begin(115200);
 delay (100);

  I2Cone.begin(27,14,100000);

  // Set the pinModes
  pcf8574.pinMode(P0, OUTPUT);
  pcf8574.pinMode(P1, OUTPUT);
  pcf8574.begin();
}

void loop(){
  pcf8574.digitalWrite(P0, HIGH);
  pcf8574.digitalWrite(P1, LOW);
  delay(1000);
  pcf8574.digitalWrite(P0, LOW);
  pcf8574.digitalWrite(P1, HIGH);
  delay(1000);
}

xreef, 2020-01-24:
Hi,
Sorry I respond to you on my site, and I don't remeber to write here, I repeat.

I think that the example I insert in the library is not so clear so you must change 
PCF8574 pcf8574(&I2Cone, 0x38);
in
PCF8574 pcf8574(&I2Cone, 0x38, 27, 14);
and remove
I2Cone.begin(27,14,100000);

Tell me if It’s ok.

Bye Renzo

WTHfloyd, 2020-01-24:
Hi Renzo 

Thanks - let me try.
And since I could not figure it out...

I ended up using the MCP23017 gpio extender....
I like it better.
No weird reverse logic for the status of pins.

Cheers.

Phil

closed 2020-02-07 by xreef