xreef/PCF8575_library #4: DigitalWrite  not work

ducthang123, 2020-10-09:
My code 

PCF8575 pcf8575(0x20);

long in =0;
void setup() {
 
  // put your setup code here, to run once:
  
  pcf8575.pinMode(P0, OUTPUT);
  pcf8575.begin();

 pcf8575.digitalWrite(P0, HIGH);
  in = millis();

}
void loop() {
 
 if(Serial.available())
 {
   int valA = Serial.parseInt();
   Serial.println(valA);
   
     
   if( valA ==  2){
         pcf8575.digitalWrite(P0, HIGH);
  
   }
   
   Serial.flush();
 }
 
  if(millis() - in >= 1000){

    // put your main code here, to run repeatedly:
   
    uint8_t val =  pcf8575.digitalRead(P0);
    if( val == LOW) {
      Serial.print("P0: ");
      Serial.print( "LOW");
     Serial.println();
     }
  
    in = millis();

  }
}

Although  I  write pcf8575.digitalWrite(P0, HIGH), it always prints LOW.

xreef, 2020-10-15:
Hi ducthang123,
in this library on OUTPUT pin you can only write not read.
Sorry Renzo

closed 2020-10-15 by xreef

ducthang123, 2020-10-17:
I try to turn on LED by pcf8575.digitalWrite(P0, HIGH) but It does not work.  This library can not write or read anything.  

my solution use to another library. this is   https://www.arduino.cc/reference/en/libraries/pcf8575/

xreef, 2020-10-17:
Hi duchang123,
you can't light up a led with pcf8575, with this or other library, the output amperage It's to low.
I explain this thing on my site, and I show a simple solution, and It's well descibed in the datasheet, I advise you to read the article or datasheet.
Bye Renzo

GEMI, 2023-06-08:
I am sorry but what article and datasheet are you referring to? Can you share the link to article?