xreef/PCF8575_library #6: pin labels

voblaunsane, 2021-03-24:
Hi, great libray, thank you!

 PCF8575.h define device pins as P0 - P15, see lines 68 - 83 contain:
```
#define P0  	0
#define P1  	1
#define P2  	2
#define P3  	3
#define P4  	4
#define P5  	5
#define P6  	6
#define P7  	7
#define P8  	8
#define P9  	9
#define P10  	10
#define P11  	11
#define P12  	12
#define P13  	13
#define P14  	14
#define P15  	15
```

but in real world PCF8575 device pins are labeled P00 - P07 and then P10 - P17, would you consider changing the code to:
```
#define P00  	0
#define P01  	1
#define P02  	2
#define P03  	3
#define P04  	4
#define P05  	5
#define P06  	6
#define P07  	7
#define P10  	8
#define P11  	9
#define P12  	10
#define P13  	11
#define P14  	12
#define P15  	13
#define P16  	14
#define P17  	15
```

so P0 -> P00, P1 -> P01 .. P7 -> P07, then there is no P8 and P9, insted 8th pin is P10, 9th pin is P11, etc...

not sure if you'd want to commit to this change. Posting this in hopes someone will find this usefull as it may be confusing...

xreef, 2021-06-27:
Hi voblaunsane,
sorry I lost this message, thanks for your support, I'm going to add a define to enable this set of labelling.
Sorry again and Thanks Renzo

xreef, 2021-06-27:
I add a define for backward compatibility, uncomment
`//#define NOT_SEQUENTIAL_PINOUT
`
on PCF8575.h .
New versione released. 
Thanks

closed 2021-06-27 by xreef