xreef/PCF8575_library #9: Digital write on ESP32 with I2C 16,17

podaen, 2022-01-23:
I have the digitalread on all pins working on my esp32, I followed the example but is not working. When I measure the pin to ground it stays low on digitalwrite.

This is the expansion board I am using
[image]

I have solderd vcc to vdd (for not using the voltage regulator)
[image]

In the library I did

#define NOT_SEQUENTIAL_PINOUT

```
#include <PCF8575.h>
#include <Wire.h>

TwoWire I2Cone = TwoWire(0);
//TwoWire I2Ctwo = TwoWire(1);
PCF8575 PCF(&I2Cone, 0x20, 16,17);

void setup() {
	Serial.begin(115200);
	while (!Serial);
	//PCF.pinMode(P00, INPUT);

	I2Cone.begin(16, 17, 400000); // SDA pin 16, SCL pin 17, 400kHz frequency

	for (int i = 0; i < 8; i++) {
		PCF.pinMode(i, OUTPUT);
	}
	
	//PCF.begin(16, 17);

	PCF.begin();

	write();
}
void loop() {
	delay(1000);
	//read();
}
void read() {
	Serial.print("val: ");
	Serial.println(PCF.digitalRead(P00));
}
void write() {
	//PCF.digitalWrite(P07, HIGH);

	for (int i = 0; i < 8; i++) {
		PCF.digitalWrite(i, HIGH);
	}
}
```

podaen, 2022-01-24:
Here is my log

Opening port
Port open
�Set pin 0 as 2
writeMode: 1readMode: 0
Write mode: 1
Set pin 1 as 2
writeMode: 11readMode: 0
Write mode: 11
Set pin 2 as 2
writeMode: 111readMode: 0
Write mode: 111
Set pin 3 as 2
writeMode: 1111readMode: 0
Write mode: 1111
Set pin 4 as 2
writeMode: 11111readMode: 0
Write mode: 11111
Set pin 5 as 2
writeMode: 111111readMode: 0
Write mode: 111111
Set pin 6 as 2
writeMode: 1111111readMode: 0
Write mode: 1111111
Set pin 7 as 2
writeMode: 11111111readMode: 0
Write mode: 11111111
Set write mode
usedPin pin 11111111111111111111111100000000
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 1 for pin 0 bin value 1 value 1
--> 1
1
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 11 for pin 1 bin value 10 value 1
--> 3
3
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 111 for pin 2 bin value 100 value 1
--> 7
7
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 1111 for pin 3 bin value 1000 value 1
--> 15
15
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 11111 for pin 4 bin value 10000 value 1
--> 31
31
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 111111 for pin 5 bin value 100000 value 1
--> 63
63
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 1111111 for pin 6 bin value 1000000 value 1
--> 127
127
0
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Write data 11111111 for pin 7 bin value 10000000 value 1
--> 255
255
0
Start end trasmission if stop here check pullup resistor.

xreef, 2022-01-27:
Hi @podaen,
I don't know if It's the problem you explain, but if you use
`PCF8575 PCF(&I2Cone, 0x20, 16,17);`

you don't do the 
`	I2Cone.begin(16, 17, 400000); // SDA pin 16, SCL pin 17, 400kHz frequency`

Bye renzo

closed 2022-02-20 by xreef

podaen, 2022-02-20:
Sorry I diden't reported, but I have it working now. The chip can not be sinkt. The voltage was between vcc and PO0.