xreef/PCF8575_library #12: Keypress Wiring Not registering low

neliel123, 2023-05-06:
Hi,
I tried the keypress on a switch button and followed the wiring on your home page.
I then used the code here https://github.com/xreef/PCF8575_library/blob/master/examples/keyPressedPin1/keyPressedPin1.ino
I am using an ESP32 and it actually does not work for me.

If I removed the 10K pull-down resistor connected to the ground then it now work.
I brought out my tester and I noticed that when the switch is open the reading at the P1 pin is around 4.2V (HIGH) but when I pressed the button it goes down to 3.9V which still registers as a HIGH which is not registering a LOW.

I tried removing the 10K resistor and everything is ok?

Can you comment on this or if the wiring is correct? Or I could be missing some setup on my end.  Thank you.

neliel123, 2023-05-06:
I think my solution is not appropriate... If I removed the 10K resistor then it looks like that I am shorting the VCC pin to the ground which is bad I believe.

But if I do not do that then it is not registering a LOW on the P1 input pin.

Is my concern valid or I am just making something wrong on my circuit?
I know your diagram uses an Arduino Uno board but I think they are applicable to ESP32 right?

donskytech, 2023-05-07:
Found the issue. I think the wiring on the switch on the project home page is wrong.  Change the input pin to the other end of the switch.