xreef/PCF8575_library #14: Ports as inputs not work at all

viktak, 2023-12-12:
Hi,
First of all, thanks for all your hard work on this and the PCF8574. That one I use all the time with no problems. Recently, I tried to do the same things with the PCF8575. All the ports work as expected as OUTPUT like this:
```
void setup(){
    Wire.begin(SDA_GPIO, SCL_GPIO);

    for (size_t i = 0; i < 16; i++)
        pcf.pinMode(i, OUTPUT);

    for (size_t i = 0; i < 16; i++)
        pcf.digitalWrite(i, HIGH);

    pcf.begin();
}

void loop(){
    if (millis() - i2cMillis > 250)
    {
        for (size_t i = 0; i < 16; i++)
            pcf.digitalWrite(i, 1 - pcf.digitalRead(i));

        i2cMillis = millis();
    }
}
```
Awesome! But when I try to use the ports as input they fail miserably:
```
void setup(){
    Wire.begin(SDA_GPIO, SCL_GPIO);

    for (size_t i = 0; i < 16; i++)
        pcf.pinMode(i, INPUT);

    pcf.begin();
}

void loop(){
    PCF8575::DigitalInput di = pcf.digitalReadAll();
    SerialMon.printf("%u - %u - %u - %u - %u - %u - %u - %u\r\n", di.p0, di.p1, di.p2, di.p3, di.p4, di.p5, di.p6, di.p7);
}
```
resulting in this:
```
0 - 0 - 0 - 0 - 0 - 0 - 0 - 0
```
Looking at the circuit with an oscilloscope I can see that although all the pins are pulled up to 3.3V with a 10k resistor, the PCF8575 pulls them down constantly to Gnd.

I have already spent 2 days on trying to figure out what went wrong, but I'm stuck. Any idea what I'm missing?

Thanks for any pointers!

ps: my environment is Windows 11 64 bit, VS Code with PlatformIO. and this is the relevant bit of `platformio.ini`:
```
##################################################
#   ENVIRONMENT
##################################################

[env:esp32s3]
board = esp32-s3-devkitc-1

platform = espressif32
framework = arduino

##################################################
#   PRE-BUILD
##################################################
extra_scripts = 
    pre:../../scripts/preIncrementBuildNumber.py
    platformio_upload.py

##################################################
#   BUILD
##################################################

build_type = release

board_build.filesystem = littlefs

lib_deps =
    ; xreef/PCF8574 library @ ^2.3.6    
    xreef/PCF8575 library @ ^1.1.1

##################################################
#   UPLOAD
##################################################

; upload_protocol = esptool
; upload_port = COM5
; upload_speed = 921600

```

mariusmotea, 2024-03-03:
I have the same issue, but not sure how they worked in some previews tests. Looks like when are declared as inputs the pins output LOW.

mariusmotea, 2024-03-03:
Looks like they required some external pull-up resistors in order to work.

viktak, 2024-03-03:
@mariusmotea Did you actually try it with pull up resistors? I'm asking, because I did, and it did not make a difference..

closed 2024-04-23 by xreef